open-rdc/RobotGuidance #3: chainerで簡単なサンプルを学習させる

yasuohayashibara, 2017-08-10:
以下の５つの状態の画像を50枚取得して，学習できるかを検証する

人が以下の位置にいる（もしくはいない）
１）右
２）左
３）真ん中
４）近く
５）いない

yasuohayashibara, 2017-08-10:
画像を取得，64x48に解像度を下げてからラベルをつけて学習させる．

yasuohayashibara, 2017-08-10:
画像の形式の例

http://qiita.com/tommyfms2/items/c3fa0cb258c17468cb30

http://www.casleyconsulting.co.jp/blog-engineer/python/deeplearning%E7%94%A8%E3%83%95%E3%83%AC%E3%83%BC%E3%83%A0%E3%83%AF%E3%83%BC%E3%82%AF-chainer-%E3%82%92%E8%A9%A6%E3%81%97%E3%81%A6%E3%81%BF%E3%81%9F_%E3%81%9D%E3%81%AE2/

Pethon, 2017-08-10:
import matplotlib.pyplot as plt
import numpy as np
#np.set_printoptions(threshold=np.inf)
#from sklearn.datasets import fetch_mldata
import chainer
from chainer import cuda, Variable, optimizers
import chainer.functions  as F
import sys
import os
import os.path
import cv2
import random
import argparse
import cPickle
import six
import six.moves.cPickle as pickle
from chainer import computational_graph as c

parser = argparse.ArgumentParser(description='Training Program')
parser.add_argument('--gpu', '-g', default=-1, type=int,
                    help='GPU ID (negative value indicates CPU)')
parser.add_argument('--model', '-m', default=0, help ='Trained Model')
args = parser.parse_args()

plt.style.use('ggplot')

N = 400
N_test = 100
category = 5

batchsize = 5
n_epoch   = 20
n_units   = 3000

images = []
labels = []
images_train = []
images_test = []
labels_train = []
labels_test = []

def load_images_from_folder(folder):
    images = []
    for filename in os.listdir(folder):
        img = cv2.imread(os.path.join(folder,filename))
        if img is not None:
            images.append(img)
    return images

load_images_from_folder('training')

labels = np.loadtxt(open('labels.csv',"rb"),delimiter=",")
labels = np.array(labels).astype(np.int32)

num = []
rand_num = []
for i in range(1,N_test + 1):
	num.append(i)
rand_num = random.sample(num, k = N_test / category)
rand_num = sorted(rand_num, reverse=True)

for i in range(0,N + 1,(N + N_test) / category):
	for j in rand_num:
		images_test.append(images[i+j])
		labels_test.append(labels[i+j])
		del images[N - i + j - 1]
		del labels[N - i + j - 1]

images_train = images
labels_train = labels

images_train = np.array(images_train).astype(np.float32).reshape((len(images_train), 3, 64, 48)) / 255
images_test = np.array(images_test).astype(np.float32).reshape((len(images_test), 3, 64, 48)) / 255
labels_train = np.array(labels_train).astype(np.int32)
labels_test = np.array(labels_test).astype(np.int32)

#========================================================

# --- モデルの設定 ---
if args.model != 0:
    if args.gpu >= 0:
        cuda.init(args.gpu)
    model = pickle.load(open(args.model,'rb'))
else:
    model = chainer.FunctionSet(conv1= F.Convolution2D(3, 16, 5, pad=2),
                                conv2= F.Convolution2D(16, 32, 5, pad=2),
                                l3= F.Linear(6272, 256),
                                l4= F.Linear(256, 5))
    if args.gpu >= 0:
        cuda.init(args.gpu)
        model.to_gpu()
 
# 伝播の設定
def forward(x_data, y_data, train=True):
    x, t = chainer.Variable(x_data, volatile=not train), chainer.Variable(y_data, volatile=not train)
    h = F.max_pooling_2d(F.relu(model.conv1(x)), ksize = 5, stride = 2, pad =2)
    h = F.max_pooling_2d(F.relu(model.conv2(h)), ksize = 5, stride = 2, pad =2)
    h = F.dropout(F.relu(model.l3(h)), train=train)
    y = model.l4(h)
 
    return F.softmax_cross_entropy(y, t), F.accuracy(y, t)

# オプティマイザーの設定
optimizer = optimizers.Adam()
optimizer.setup(model.collect_parameters())
 
# 学習とテスト
for epoch in six.moves.range(1, epochNumber + 1):
    print('epoch', epoch)
    trainImageNumber = len(labels_train)
    testImageNumber = len(labels_test)
 
    perm = np.random.permutation(trainImageNumber)
 
    # 学習
    sumAccuracy = 0
    sumLoss = 0
    for i in six.moves.range(0, trainImageNumber, batchNumber):
        x_batch = images_train[perm[i:i + batchNumber]]
        y_batch = labels_train[perm[i:i + batchNumber]]
 
        if args.gpu >= 0:
            x_batch = cuda.to_gpu(x_batch)
            y_batch = cuda.to_gpu(y_batch)
 
        optimizer.zero_grads()
        loss, acc = forward(x_batch, y_batch)
        loss.backward()
        optimizer.update()
 
        if epoch == 1 and i == 0:
            with open("graph.dot", "w") as o:
                o.write(c.build_computational_graph((loss, )).dump())
            with open("graph.wo_split.dot", "w") as o:
                g = c.build_computational_graph((loss, ),
                o.write(g.dump())

            print('graph generated')

        sumLoss += float(loss.data) * len(y_batch)
        sumAccuracy += float(acc.data) * len(y_batch)
 
    print('train mean loss={}, accuracy={}'.format(
        sumLoss / trainImageNumber, sumAccuracy / trainImageNumber))
 
    # テスト
    sumAccuracy = 0
    sumLoss = 0
    for i in six.moves.range(0, testImageNumber, batchNumber):
        x_batch = images_test[i:i + batchNumber]
        y_batch = labels_test[i:i + batchNumber]
        if args.gpu >= 0:
            x_batch = cuda.to_gpu(x_batch)
            y_batch = cuda.to_gpu(y_batch)
 
        loss, acc = forward(x_batch, y_batch, train=False)
 
        sumLoss += float(loss.data) * len(y_batch)
        sumAccuracy += float(acc.data) * len(y_batch)
 
    print('test mean loss={}, accuracy={}'.format(
        sumLoss / testImageNumber, sumAccuracy / testImageNumber))
 
    #学習後にモデルを保存
    pickle.dump(model, open('model'+str(epoch), 'wb'),-1)

yasuohayashibara, 2017-08-27:
sampleを用いた学習は確認できているので，issueを閉じます．

closed 2017-08-27 by yasuohayashibara