open-rdc/RobotGuidance #19: 2017/09/21の実験結果の解析

Pethon, 2017-09-22:
報酬の移動平均のグラフ（区間１０００）
[image]

2200回目から報酬が下がり始めたことを確認できました。
報酬の遅れ問題も確認したが、問題なく正常に0.2秒の周期でした。

報酬の平均周期　0.200369782
報酬の最長周期　0.203059968

Pethon, 2017-09-22:
2017/09/21
実験を行った最中に、パソコンの電源が落ちてしまったため、
bagファイルをちゃんと保存できず、中身のlogを見られていません。
bagファイルは下のリンクにアップロードしました。
[https://drive.google.com/open?id=0B6WmN6afJ0WJM25OYjdHSG1VcjQ](url)
rosbag reindex と
rosbag fix というコマンドで回復できるそうですが、うまく行けませんでした。

cpuの温度は学習中に熱くないと思います。
電源が落ちる前に、温度が少し上がったが、学習のスピードに影響しないと思います。
cpuのグラフです。
[image]

Pethon, 2017-09-23:
2017/09/23
実験結果の動画
[https://youtu.be/w3dGtF3V81o](url)

Pethon, 2017-09-24:
2017/09/24
報酬の生データと移動平均した値。
[https://drive.google.com/open?id=0B6WmN6afJ0WJZTZtRGNCdG9lb28](url)

報酬の移動平均のグラフ（区間２０００）
[image]

yasuohayashibara, 2017-10-03:
SI2017の予稿集を提出して解析が一段落したので，issueを閉じます．

closed 2017-10-03 by yasuohayashibara