open-rdc/RobotGuidance #24: 前進の行動を選択しない問題

Pethon, 2018-08-23:
実ロボットを用いて実験を行うと，ロボットが前進しないことが多い．
ロボットがより高い報酬がもらえる行動を探索しきれない理由は，正の報酬を与えているからだと考えられる．
シミュレーター上で，正の報酬は，ロボットが前進の行動を選択しないことに関係あるかを確認する．

ロボットの行動：前進，右旋回，左旋回，停止
報酬：正の値
最高の報酬：2.0
学習回数：1万ステップまで
緑の円が人間とする

ロボットが旋回すると，緑の円が左右に動く．
ロボットが前進すると，緑の円のサイズが大きくなる．
高い報酬を習得できる条件は，緑の円が真ん中にあって，ちょうどいいサイズになる状態という条件
環境が100回ごとにリセットする（緑の円をランダムなサイズと場所に配置する）

yasuohayashibara, 2018-08-23:
動画のリンクが切れています．

Pethon, 2018-08-23:
@yasuohayashibara 6000回までの報酬です。（充電しないといけないので、一旦休憩です）
[image]

yasuohayashibara, 2018-08-23:
シミュレーションのように一様に上昇していくというわけではなさそうですね．

Pethon, 2018-08-23:
シミュレーターより複雑なので、収束はこれからかもしれません。

Pethon, 2018-08-23:
@yasuohayashibara 
6800ステップだけ学習したが，学習中には結構追従できたので，試しで本番の追従をさせてみました．
9:30からが追従フェーズです．
普通の速度で再生します．
ロボットとの距離がだいたい30-50cmだと思います．（１マス目〜1.5マス目）
[https://youtu.be/NZRAIXqeHEg](url)

yasuohayashibara, 2018-08-23:
動画を見る限り追従及び停止ができているように思います．
本日，申し込みに関する議論ができればと思います．

Pethon, 2018-08-26:
負の報酬を与えるとロボットがちゃんと探索してくれるので，前進しない問題を解けました．

closed 2018-08-26 by Pethon