open-rdc/RobotGuidance #27: エピソードを切替の効果

Pethon, 2018-08-26:
実験結果

実験1
[image]

実験2
[image]

実験3
[image]

実験4
[image]

実験5
[image]

実験6
[image]

実験7
[image]

実験8
[image]

実験9
[image]

実験10
[image]

Pethon, 2018-08-26:
@yasuohayashibara 実験結果から見ると，エピソード切り替えてもあまり効果がないです．（バッグはもうないと思いますが，念の為にもう一度確認していただきたいです．）
考えられる理由は，今の割引率が0.95ですので，0.95^100は0.006ぐらいです．学習にあまり影響しないと思います．
利用しても，学習は早くならないが，人間の挙動による誤学習を防止できるぐらいだと思います．

yasuohayashibara, 2018-08-27:
こちらの計算に使用したプログラムのリンクをお願いします．

Pethon, 2018-08-27:
@yasuohayashibara 
この２つのファイルです．
[https://github.com/open-rdc/RobotGuidance/tree/master/agent](url)
[https://github.com/open-rdc/RobotGuidance/blob/master/ros/robot_guidance/robot_guidance/scripts/robot_guidance_node.py](url)

agentはこちらで
[https://github.com/open-rdc/RobotGuidance/tree/master/agent](url)

yasuohayashibara, 2018-08-30:
チェックが遅れており申し訳ありません．
まだ動作確認しただけで何とも言えないので，もう少し検討したいと思います．

yasuohayashibara, 2018-08-31:
@Pethon  ３つめの引数の意味を知っていたら教えて下さい．
self.action = self.rl.stop_episode_and_train(imgobj, self.reward, self.done)

Pethon, 2018-08-31:
@yasuohayashibara 
doneはこのエピソードが本当に終わるかどうかを確認するだけだと思います。
[https://chainerrl.readthedocs.io/en/latest/_modules/chainerrl/agents/reinforce.html#REINFORCE](url)
このサイトによると、done=falseの場合には、このエピソードを無視すると書いてあります。
未完成のエピソードを誤学習しないように確認していると思います。

yasuohayashibara, 2018-09-29:
あまり効果がないという結論でしたので，issueを閉じます．
必要に応じて再度オープンします．

closed 2018-09-29 by yasuohayashibara