open-rdc/RobotGuidance #28: 状態によって選択される行動のグラフをリアルタイムに出力する

yasuohayashibara, 2018-09-02:
状態によって選択される行動のグラフをリアルタイムに出力する

yasuohayashibara, 2018-09-02:
こちらでコミットした．
https://github.com/open-rdc/RobotGuidance/commit/711286db0e78a9e334460a6cc2651819490015e3

以下で実行
oslaunch robot_guidance_check_display_action.launch

結果
https://youtu.be/hvHdGUZih2g

結果的に100stepごとに各状態に対する行動を求めているが，あまりにも急激に変化しており，何か根本的に問題がありそうな気がする．学習をやめたタイミングで結果が大きく変化する．

yasuohayashibara, 2018-09-02:
報酬をプラスにした時の挙動
https://youtu.be/F1p2aP3dt00

```
        self.reward_lr = self.reward_lr ** 3
        self.reward_fb = self.reward_fb ** 3
        self.reward = self.reward_lr + self.reward_fb
```

@Pethon 
まずは予稿集を書くのが最優先ですが，状態に対する行動の変化があまりにも急激ですので，その原因を探ってください．これが正しいとすると学習を打ち切ったタイミングで，追従できるかどうかが決まります．
学習係数などの調整が必要かもしれません．

Pethon, 2018-09-12:
３つの行動で実験してみました．
`self.reward_lr = self.reward_lr ** 3
        self.reward_fb = self.reward_fb ** 3
        self.reward = self.reward_lr + self.reward_fb`
[https://youtu.be/k6XwVTxyQTs](https://youtu.be/k6XwVTxyQTs)

`self.reward_lr = self.reward_lr ** 3 - 1
        self.reward_fb = self.reward_fb ** 3 - 1
        self.reward = self.reward_lr + self.reward_fb`
[https://youtu.be/BvnbUIOM85k](https://youtu.be/BvnbUIOM85k)

yasuohayashibara, 2018-09-13:
@Pethon 動画を全て見るのはつらいので，コメントもほしいです．

yasuohayashibara, 2018-09-13:
結論として，rewardがマイナスの場合は学習が収束しないということで良いでしょうか．

Pethon, 2018-09-13:
@yasuohayashibara 今のところは学習中に，正の報酬でも，負の報酬でも学習が収束したが，テストするとき，負の報酬だけは適切な行動を選択してくれません．

Pethon, 2018-09-13:
上田先生と相談して，この問題がやはりバッグがあると思われます．

yasuohayashibara, 2018-09-14:
どんなバグ？

Pethon, 2018-09-14:
@yasuohayashibara まだわかりません．

yasuohayashibara, 2018-09-14:
どのような現象がバグと判断されたかという質問です．
どのような状態になればバグが無くなったと判断できるかを定義しないと改善の方針を立てることができません．

Pethon, 2018-09-18:
同じ画像に対して，以下の2つの関数を使ってみて，同じ行動を獲得できるかを確認する．
```
self.rl.act_and_trains(imgobj, self.reward)
self.rl.act(imgobj)
```

yasuohayashibara, 2018-09-20:
上記検証しました？

Pethon, 2018-09-20:
@yasuohayashibara 検証しました．特に異常なしです．
違う行動を選択する場合はありますが，それはepsilonを利用しているからです．
[https://youtu.be/XXsW0WtAts8](https://youtu.be/XXsW0WtAts8)

yasuohayashibara, 2018-09-20:
お疲れさま．
これで分かったのは，１試行ごとに出力が大きく変化するような強化学習器になっているということですね．
ミーティングで議論したいので，上記の結果を分かりやすくプレゼンする準備をしておいて下さい．

Pethon, 2018-09-21:
10 steps = 1 episodeと，1episode学習したらテストするという条件で実験しました．
テストするときに毎回獲得した行動が異なる部分があるので，やはりおかしいです．（ターゲットネットワークの更新は100ステップごとに行うので，更新しない限りには同じ行動が出るはず．）
[https://youtu.be/QJ_RjXuyAjs](https://youtu.be/QJ_RjXuyAjs)

Pethon, 2018-09-21:
上田先生ともう一度相談しました．
各ステップで学習を行うと価値関数が変わるので，テストするときに獲得できる行動が変わると言いました．

Pethon, 2018-09-21:
1回学習　→　エピソードをストップ　→　テストのループで実験しました．
テストするときに，急激に変化せず，少しつづ変化することが確認できました．
[https://youtu.be/eu_bcwarVoM](https://youtu.be/eu_bcwarVoM)
というわけで，各ステップに価値関数が変わることによってテストで獲得する行動が少しつづ変わる．
さらに，１００回学習すると，価値関数が結構変わるため，テストの行動が急激に変化すると思います．

ソースコードがfix_#28のブランチにあります．
[https://github.com/open-rdc/RobotGuidance/tree/fix_%2328](https://github.com/open-rdc/RobotGuidance/tree/fix_%2328)

yasuohayashibara, 2018-09-22:
> 10 steps = 1 episodeと，1episode学習したらテストするという条件で実験しました．
> テストするときに毎回獲得した行動が異なる部分があるので，やはりおかしいです．（ターゲットネットワークの更新は100ステップごとに行うので，更新しない限りには同じ行動が出るはず．）
> https://youtu.be/QJ_RjXuyAjs

確かに以下でtarget_update_intervalは100になっているのに，それ以下で選択される行動が変化するのはおかしいですね．

https://github.com/open-rdc/RobotGuidance/blob/cbf37f191170434d35fe5da982bda10095721cd6/ros/robot_guidance/robot_guidance/scripts/reinforcement_learning.py#L45-L48

Pethon, 2018-09-22:
DoubleDQNのソースコード
[https://chainerrl.readthedocs.io/en/latest/_modules/chainerrl/agents/double_dqn.html#DoubleDQN](https://chainerrl.readthedocs.io/en/latest/_modules/chainerrl/agents/double_dqn.html#DoubleDQN)
DQNを利用しているので，DQNのソースコードも貼っておきます．
[https://chainerrl.readthedocs.io/en/latest/_modules/chainerrl/agents/dqn.html#DQN](https://chainerrl.readthedocs.io/en/latest/_modules/chainerrl/agents/dqn.html#DQN)

yasuohayashibara, 2018-09-22:
こちらがtarget_networkを更新する関数になると思いますが，どのタイミングで呼ばれるかを調査してもらえますか？

https://github.com/chainer/chainerrl/blob/590647a58915b3d7ed462ca34acfee13a0cc03ee/chainerrl/agents/dqn.py#L392

Pethon, 2018-09-22:
dqn.pyは　/usr/local/lib/python2.7/dist-packages/chainerrl/agents/dqn.py　にあります．

yasuohayashibara, 2018-09-22:
設定通りtarget_networkが100回に1回更新することをパシン君が確認しました．

Pethon, 2018-09-24:
chainerのAdamがおかしいという記事を見つけましたが，chainerのソースコードをまだ確認していません．
[https://qiita.com/tokkuman/items/1944c00415d129ca0ee9#adam](https://qiita.com/tokkuman/items/1944c00415d129ca0ee9#adam)

Pethon, 2018-09-26:
シミュレーターで50000回学習を行いました。
[image]
[image]
[image]

Pethon, 2018-09-26:
実機の実験結果(一番良く選択された行動のみ)
画像は論文を書き直せたら作ります．
100 :
↓		↓		↓		↓		↓
↓		↓		↓		↓		↓
↓		↓		↓		↓		↓
↓		↓		↓		↓		↓
				人

200 :
↓		↓		↓		↓		↓
↓		↓		↓		↓		↓
↓		↓		↓		↓		↓
↓		↓		↓		↓		↓
				人

2000 :
ｘ		↓		↓		→		ｘ
ｘ		ｘ		↓		→		ｘ
ｘ		←		↓		ｘ		ｘ
ｘ		ｘ		ｘ		ｘ		ｘ
				人

2100 :
ｘ		←		←		→		→
ｘ		←		←		→		ｘ
ｘ		←		←		→		ｘ
ｘ		ｘ		←		ｘ		ｘ
				人

2200 :
ｘ		↓		ｘ		→		ｘ
ｘ		↓		ｘ		→		ｘ
ｘ		↓		←		→		ｘ
ｘ		ｘ		ｘ		ｘ		ｘ
				人

3000 :
←		←		←		→		→
→		←		←		→		←
←		←		←		→		←
←		→, ←	→		←		←
				人

3100 :
←		←		←		→		→
→		←		←		→		←
→		←		←		→		←
→		→		→		←		→
				人

3200 :
←		←		←		ｘ		ｘ
ｘ		←		←		ｘ		←
←		←		←		ｘ		←
←		ｘ		ｘ		←		←
				人

4000 :
→		→		→		→		→
→		↓		→		→		→
→		↓		→		→		→
→		→		→		→		→
				人

4100 :
→		→		ｘ		→		→
→		↓		ｘ		→		→
→		↓		ｘ		→		→
→		↓		→		→		→
				人

4200 :
←		←		←		ｘ		ｘ
ｘ		←		←		ｘ		ｘ
←		←		←		ｘ		←
←		←		←		ｘ		←
				人

5000 :
ｘ		ｘ		ｘ		→		→
ｘ		ｘ		ｘ		→		ｘ
ｘ		ｘ		ｘ		→		ｘ
ｘ		ｘ		→, ｘ	ｘ		ｘ
				人

5100 :
ｘ		→		→		→		→
→		←		→		→		ｘ
→		ｘ		↓		→		ｘ
→		ｘ		→		→		→
				人

5200 :
↓		↓		←		→		→
↓		←		←		→		↓
↓		↓		↓		→		↓
↓		↓		→		ｘ		↓
				人

6000 :
ｘ		←		↓		ｘ		ｘ
ｘ		←		↓		→		ｘ
ｘ		↓		↓		→		ｘ
ｘ		ｘ		→		ｘ		ｘ
				人

6100 :
←		←		←		→		→
←		←		↓		→		←
←		←		↓		→		←
←		←		←		ｘ		←
				人

6200 :
ｘ		←		←		→		→
→		←		↓		→		↓
ｘ		↓		↓		→		ｘ
ｘ		ｘ		→		ｘ		ｘ
				人

報酬グラフ
[image]

yasuohayashibara, 2018-09-27:
左右の矢印がほとんど外側に向いていますが，間違いありませんでしょうか．

Pethon, 2018-09-27:
反転するのが忘れてしまって，間違えました．
上のコメントを修正します．

yasuohayashibara, 2018-09-27:
良かったです．

Pethon, 2018-09-29:
行動グリッドだけ見るのは，ただの確率が高い行動はどれかしか分からないので，
それぞれの行動を選ぶ確率も確認する．

実験結果
56900回学習した結果
[image]

learning = True count: 56900 action: 2, reward: 0.73974 
variable([[ 0.08563501  0.18902992  0.38180795  0.34352705]])
variable([[ 0.0672768   0.14796838  0.51679593  0.26795885]])
variable([[ 0.14901774  0.25696889  0.32152379  0.27248958]])
variable([[ 0.23989956  0.17138113  0.38853985  0.20017949]])
variable([[ 0.19185238  0.20963034  0.34492496  0.25359231]])
variable([[ 0.2650536   0.19689037  0.3078666   0.23018944]])
variable([[ 0.32164323  0.24069938  0.22439161  0.21326573]])
variable([[ 0.21099417  0.34171849  0.19697228  0.25031507]])
variable([[ 0.23280178  0.34785786  0.17289788  0.24644242]])
variable([[ 0.42328849  0.28509617  0.14623506  0.14538018]])
variable([[ 0.3052578   0.43834105  0.14835343  0.1080478 ]])
variable([[ 0.447274    0.31222257  0.15514219  0.08536121]])
variable([[ 0.07094917  0.18529727  0.50995326  0.23380025]])
variable([[ 0.12602054  0.23377664  0.35238931  0.28781351]])
variable([[ 0.21222432  0.20071414  0.36215749  0.22490408]])
variable([[ 0.23231971  0.17588633  0.34560448  0.24618952]])
variable([[ 0.23231971  0.17588633  0.34560448  0.24618952]])
variable([[ 0.32207108  0.23488647  0.22172277  0.22131968]])
variable([[ 0.19626366  0.37258434  0.17316934  0.2579827 ]])
variable([[ 0.1989613   0.3857657   0.16693613  0.24833688]])
variable([[ 0.40463781  0.27455801  0.16351403  0.15729009]])
variable([[ 0.21978565  0.49614164  0.16281787  0.12125483]])
variable([[ 0.47096077  0.33572933  0.12605275  0.06725719]])
variable([[ 0.05664492  0.18521063  0.50755519  0.25058925]])
variable([[ 0.0943212   0.19842188  0.4113813   0.29587552]])
variable([[ 0.18811429  0.25852203  0.29492539  0.25843823]])
variable([[ 0.2168411   0.1782632   0.35732102  0.24757466]])
variable([[ 0.23270917  0.15897562  0.3614004   0.24691476]])
variable([[ 0.29788756  0.22682717  0.23180796  0.24347726]])
variable([[ 0.18832193  0.37624696  0.17277502  0.26265618]])
variable([[ 0.23129278  0.32916155  0.20172486  0.23782077]])
variable([[ 0.36779326  0.25337577  0.20859024  0.17024072]])
variable([[ 0.19059606  0.5258826   0.15460728  0.12891401]])
variable([[ 0.4046413   0.42202166  0.11853709  0.05479994]])
variable([[ 0.03863584  0.15250784  0.5647729   0.24408337]])
variable([[ 0.09400778  0.19346291  0.42083615  0.29169321]])
variable([[ 0.17805192  0.25917596  0.29624182  0.26653031]])
variable([[ 0.25378302  0.16539761  0.34637117  0.23444827]])
variable([[ 0.26796022  0.14038524  0.35924467  0.23240989]])
variable([[ 0.31523368  0.20520629  0.23674847  0.24281161]])
variable([[ 0.1837561   0.38125762  0.16851668  0.26646957]])
variable([[ 0.24114758  0.30332938  0.21878104  0.23674195]])
variable([[ 0.28383896  0.23345774  0.27040201  0.21230131]])
variable([[ 0.19152157  0.54217505  0.14163743  0.12466601]])
variable([[ 0.36612543  0.48455402  0.10167512  0.04764545]])
variable([[ 0.03092476  0.13305615  0.59783047  0.23818853]])
variable([[ 0.07929399  0.17868546  0.45738983  0.28463075]])
variable([[ 0.15242244  0.23337595  0.34039199  0.27380964]])
variable([[ 0.24468862  0.15703681  0.35668799  0.24158664]])
variable([[ 0.2679067   0.13228464  0.36327854  0.23653011]])
variable([[ 0.31720766  0.19575763  0.24162467  0.24541005]])
variable([[ 0.18316258  0.37802729  0.16657972  0.27223042]])
variable([[ 0.23827247  0.29009417  0.22948907  0.24214432]])
variable([[ 0.27759588  0.24482101  0.2635926   0.21399048]])
variable([[ 0.19654374  0.515742    0.1373775   0.15033674]])
variable([[ 0.32535958  0.54356343  0.08778378  0.04329314]])
variable([[ 0.0309      0.1542846   0.56838131  0.24643406]])
variable([[ 0.05796375  0.14175174  0.52822793  0.27205655]])
variable([[ 0.10405889  0.16970995  0.45621637  0.27001479]])
variable([[ 0.14062704  0.16074516  0.40729988  0.29132792]])
variable([[ 0.11183172  0.14502054  0.42530194  0.31784576]])
variable([[ 0.12148374  0.24184361  0.28898996  0.34768271]])
[... 36 lines cut ...]
variable([[ 0.69646966  0.2093398   0.04738745  0.04680308]])
variable([[ 0.53571713  0.3690578   0.05195615  0.04326897]])
variable([[ 0.11220068  0.28373635  0.38445482  0.21960811]])
variable([[ 0.11257421  0.16087654  0.48786357  0.23868568]])
variable([[ 0.15597372  0.18509415  0.40436393  0.25456822]])
variable([[ 0.19364195  0.19159691  0.38070107  0.23406006]])
variable([[ 0.13861056  0.1466327   0.46338949  0.2513673 ]])
variable([[ 0.06974693  0.29091847  0.36772329  0.27161133]])
variable([[ 0.03968829  0.29175153  0.45664805  0.21191223]])
variable([[ 0.06752744  0.23759401  0.43501553  0.25986296]])
variable([[ 0.25769928  0.30022463  0.22743987  0.21463618]])
variable([[ 0.67405838  0.21968323  0.05514655  0.05111182]])
variable([[ 0.69645244  0.22644897  0.04019941  0.03689919]])
variable([[ 0.10632857  0.25072983  0.43569452  0.20724702]])
variable([[ 0.09125376  0.14775915  0.53709698  0.2238901 ]])
variable([[ 0.12683429  0.13890895  0.47276121  0.26149553]])
variable([[ 0.19021316  0.14020233  0.41359016  0.25599435]])
variable([[ 0.11247382  0.12696022  0.50942147  0.25114453]])
variable([[ 0.05273364  0.17663111  0.57143813  0.1991972 ]])
variable([[ 0.03450513  0.23465969  0.54213655  0.1886986 ]])
variable([[ 0.05151573  0.23166485  0.46789423  0.24892525]])
variable([[ 0.24326807  0.28799853  0.26878563  0.19994776]])
variable([[ 0.5748387   0.26246136  0.08634965  0.07635032]])

明らかに，ピークになった時もありますが，差がそれほど高くない時もあります．

yasuohayashibara, 2018-09-29:
こんな感じのグラフにして下さい．

https://www.rinearn.com/graph3d/image/performance1.jpg

Pethon, 2018-10-02:
181001の実験結果
[https://youtu.be/GsJgux6pYLc](https://youtu.be/GsJgux6pYLc)
人間はプレートの左下にいます（xの文字のところです）
x = 0; ロボットが左側にある
x = 10; ロボットが右側にある
y = 0; 距離が近い
y = 8; 距離が遠い

グリッド図は作成中です．

yasuohayashibara, 2018-10-02:
長時間の実験お疲れさまです．
左右は波形にはっきり現れますね．
前進は上がったり下がったりを繰り返しています．
良い実験結果だと思いますが，近づいて停止というのはできるようになりましたでしょうか？

Pethon, 2018-10-02:
@yasuohayashibara 15000回やっても前後はなかなか学習してくれないです．
人が下がったら，すぐに進んでくれないことが多くて，下がり過ぎたら全然進まなくなりました．（この前に良くある停止が多い問題です）

Pethon, 2018-10-02:
181002の実験結果
報酬グラフ
[image]

行動グラフ
[https://youtu.be/N5hzNAP4y8g](https://youtu.be/N5hzNAP4y8g)

行動グリッド
[https://youtu.be/BvQKVkecPEE](https://youtu.be/BvQKVkecPEE)

yasuohayashibara, 2019-05-02:
すでに解決しているのでissueを閉じます．

closed 2019-05-02 by yasuohayashibara